OpenKNX/OAM-PresenceModule #39: Kurzzeit per KO umgehen.

traxanos, 2022-11-08:
Ich würde gerne im HWR die Kurzzeit während die Waschmaschine oder Trockner läuft bzw. eingeschaltet ist, die Kurzzeit aussetzen. Der HWR ist auch ein Durchgangszimmer, daher wäre eine Kurzzeit praktisch, sorgte aber dafür, dass die Frau dauernd im dunkeln sitzt. Daher haben wir das wieder ausgeschaltet.

mumpf, 2022-11-08:
Ja klar, dafür einfach eine andere Tagesphase nehmen.

mumpf, 2022-11-08:
Oder Du bastelst Dir eine eigene Kurzzeit, indem Du die Nachlaufzeit selber über Logik von extern setzt (oder auch nicht).
Ich habe auf absehbare Zeit (bis zur nächsten größeren Überarbeitung, nicht vor nächstem Jahr) keine KO merhr frei.

traxanos, 2022-11-08:
Das extern zu lösen ist viel zu kompliziert, das steht dann in keinem Verhältnis. Das ist aber auch kein Prio-Thema.

willisurf, 2022-11-08:
und die Variante dafür einfach eine andere Tagesphase zu nehmen?

traxanos, 2022-11-11:
Ja das wäre eine Möglichkeit, aber solange die Tagesphase nur per Szene umgeschaltet werde kann, ist das für mich keine Option. Für so eine Logik möchte ich keine Szene opfern. Und jetzt ein neues Szenen-GA zu erstellen, widersprich meine Verständnis für Wartbarkeit. Dann verzichte ich lieber.

mumpf, 2022-11-11:
Das ist deine Entscheidung. 
Wie schon erwähnt, sind keine KO mehr frei. Aber unabhängig davon fehlt mir für Deine Antwort jegliches Verständnis. 
Falls es nur um ein Szenen-KO geht: Du kannst auch ein DPT 5 nehmen. 
Ansonsten werde ich nicht für dein Verständnis von Wartbarkeit programmieren.
Gruß, Waldemar

closed 2022-11-11 by mumpf

traxanos, 2022-11-11:
Das Problem bei Szene ist, dass ich Szenen bereits verwende. Sprich das Szenen-KO ist schon mit einer zentralen Szenen-GA verbunden. Ich muss also solche Sonderlocken (die wenn jeweils nur einen Raum betrifft), mit zwei zentralen Szene abbilden. Bei nur 64 Szenen ist das
a) schnell voll wie bei mir
b) schnell unübersichtlich, da dies ja nichts ist, was mehrere Geräte betrifft.

Die Alternative ist eine separate GA anzulegen und die wichtigen Szenen die auch den VPM betrifft von der zentralen GA auf die "Sub"-Szene zu kopieren. Also brauche ich dann auch auch noch eine Logik (oder zumindest freie Bereiche für eine parallel verknüpfte GA). Das ist alles möglich, aber wie gesagt für so eine minimal Funktion übertrieben.

Und ob etwas wartbar ist, entscheide ich anhand wie aufwendig es ist etwas in Text in meine Doku zu übernehmen. Und wen dann ein dritter direkt versteht was ich da gebaut habe, dann ist es wartbar für mich.

Und wenn da nichts frei ist, oder du das für nicht sinnvoll erachtet ist das für mich überhaupt kein Problem.

mumpf, 2022-11-11:
Ja, ich kann die Problematik zum Teil nachvollziehen, aber der Teil

> Die Alternative ist eine separate GA anzulegen und die wichtigen Szenen die auch den VPM betrifft von der zentralen GA auf die "Sub"-Szene zu kopieren. Also brauche ich dann auch auch noch eine Logik (oder zumindest freie Bereiche für eine parallel verknüpfte GA). 

ist nicht wirklich so kompliziert, wie Du das beschreibst. Für Tagesphasen wirst Du in Deiner globalen GA ja maximal 4 Szenen reserviert haben. Also bleiben noch 60 Szenen übrig, die man in einer weiteren GA "verbraten" kann. Und diese GA kann einfach Hörend auf dem Eingang liegen, da muss nichts kopiert werden. Es dürfen nur nicht die selben Werte sein. 

Was ich mir noch vorstellen könnte (nicht sofort, aber bei einer zuküfntigen Erweiterung): Ich erlaube neben Szenen bei diesem KO auch DPT5, dann hätte man mehr Werte - und könnte sogar die unteren 0-63 mit Szenen gleichsetzen. Würde so was helfen?

traxanos, 2022-11-11:
Jetzt glaube ich, das du was missverstehst. Ich habe kein Problem mit Szenen für den Wechsel der Tagesphase an sich! Das ist ja was Globales und genau dafür sind Szenen perfekt. Es ging darum .z.B. in Räumen abhängig von einem Event die Kurzzeit auszusetzen. Die Nutzung der Tagesphase war ja nur eine Idee von willisurf das mit Trick17 anders zu lösen.
Das bedeutet aber 2 zusätzliche Szenen je Raum (Raum X normal & Raum x Kurzzeit aus). Hast du also 4 Räume, dann macht das 8 weitere Szenen.

Ich hab aber nun eine bessere Lösung für mich gefunden. Man kann ja bei 2 Tagesphasen ein Tag/Nacht-KO nutzen. Damit kann ich dann direkt den Boolean auswerten. Perfekt!

PS: Die Idee mit dem DPT5 verstehe ich und eigentlich echt clever, aber ich denke das dies übertrieben und eigentlich auch nicht so KNX typisch ist. Dafür ist die Zeit zu kostbar :D

mumpf, 2022-11-11:
> Das bedeutet aber 2 zusätzliche Szenen je Raum (Raum X normal & Raum x Kurzzeit aus). Hast du also 4 Räume, dann macht das 8 weitere Szenen.

Und da liegst Du falsch: Das sind nur 2 Szenen und 4 GA, für jeden Raum eine. Die kommen jeweil hörend in dem jeweiligen Raum auf den PM, zusätzlich zu Deiner Haupt-Szenen-GA. Ob Du an dem Eingang Deine Haupt-Szenen-GA überhaupt bräuchtest, kann ich nicht sagen, ich habe für Tagesphasen z.B. komplett andere GA genommen.

Gruß, Waldemar

traxanos, 2022-11-12:
Das wäre aber eine andere Art der Umsetzung. Das war mir aber auch schon vorher Anfang an klar. Daher hatte ich ja am anfang schon folgendes geschrieben:

"(oder zumindest freie Bereiche für eine parallel verknüpfte GA)"