OpenKNX/OAM-PresenceModule #40: Licht schaltet nicht korrekt aus.

traxanos, 2022-11-16:
Aktuell schaltet das Licht nicht mehr aus.

GA Logger
https://knx-user-forum.de/forum/projektforen/openknx/1808556-openknx-virtualpresence-release-vpm?p=1817658#post1817658

Settings und Debuginformationen
https://knx-user-forum.de/forum/projektforen/openknx/1808556-openknx-virtualpresence-release-vpm?p=1817765#post1817765

traxanos, 2022-11-16:
Sobald ich den zyklische senden von der Türe entferne, klappt es. Sobald ich das wieder aktiviere, hab ich dauerlicht :D Deine Eingebung scheint also zu passen. Ich nehme den Sensor solange vom VPM.

mumpf, 2022-11-17:
Möglicher Workaround hier: [https://knx-user-forum.de/forum/projektforen/openknx/1808556-openknx-virtualpresence-release-vpm?p=1817996#post1817996](https://knx-user-forum.de/forum/projektforen/openknx/1808556-openknx-virtualpresence-release-vpm?p=1817996#post1817996)

mumpf, 2022-11-17:
So, der Fehler war ein guter! 
Es war wirklich so, dass wiederholte AUS bei schaltenden Präsenz- bzw. Bewegungs-KO die Nachlaufzeit nachtriggern konnten. 
Hat wohl vorher keiner getestet.

Danke, ist mit Release 1.2 gelöst.

closed 2022-11-17 by mumpf

traxanos, 2022-11-19:
```
241  19.11.2022 13:39:16,208  vom Bus    Niedrig  1.0.142  Taster Badezimmer Flur  1/1/96  EG Badezimmer Licht Taster (Kurz)  6  GroupValue_Write    $00 | Aus
246  19.11.2022 13:39:16,360  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $00 | Aus
247  19.11.2022 13:39:16,360  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $00 | Aus
249  19.11.2022 13:39:16,406  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $00 | Aus
252  19.11.2022 13:39:16,713  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $00 | Aus
265  19.11.2022 13:39:17,443  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/7/80  EG Badezimmer Reset (inv)  6  GroupValue_Write    $00 | Aus
266  19.11.2022 13:39:17,489  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer PM 1  6  GroupValue_Write    $00 | Aus
267  19.11.2022 13:39:17,535  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $01 | Ein
269  19.11.2022 13:39:17,694  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $01 | Ein
270  19.11.2022 13:39:17,737  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $01 | Ein
273  19.11.2022 13:39:18,168  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $01 | Ein
282  19.11.2022 13:39:19,351  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/83  EG Badezimmer PM 2 TP  6  GroupValue_Write    $01 | Ein
284  19.11.2022 13:39:19,396  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/82  EG Badezimmer PM 2 BW  6  GroupValue_Write    $00 | Aus
```

Auf die schnelle ein Log Auszug (Konfig kann ich später schicken).

Hab mit 241 ausgeschaltet, der VPM schaltet mit 246 aus. Der BWM sendet in 266 ein AUS (aufgrund des PM Reset) und im Anschluss geht 267 geht das Licht wieder an. Dazwischen ist nicht gekommen.

Wenn also die beiden Cherrypicks reichen für die neue Version 1.2 (was mit auch die KNX für KNXProd + FW anzeigt), dann scheint es doch nicht behoben zu sein.

traxanos, 2022-11-19:
Hier ein größerer Auszug mit Erläuterung was passiert ist:

PM 1 ist der BWM an der Wand und als Bewegung verknüpft.
PM 2 (BW/TP als OR verknüpft) ist der Steinel TPM als "nur verlängern nicht einschalten"

```
Id Zeit  Dienst  Flags   Prio  Quelladresse  Quellname  Zieladresse  Zielname  Hop Count  Typ  DPT  Info
400  19.11.2022 17:45:40,659  vom Bus    Niedrig  1.0.142  Taster Badezimmer Flur  1/1/96  EG Badezimmer Licht Taster (Kurz)  6  GroupValue_Write    $00 | Aus
401  19.11.2022 17:45:40,709  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $00 | Aus
403  19.11.2022 17:45:40,879  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $00 | Aus
404  19.11.2022 17:45:40,879  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $00 | Aus
406  19.11.2022 17:45:41,202  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $00 | Aus
410  19.11.2022 17:45:41,795  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/7/80  EG Badezimmer Licht Reset (inv)  6  GroupValue_Write    $00 | Aus
411  19.11.2022 17:45:41,957  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/73  EG Badezimmer Licht (Decke) Helligkeit Status  6  GroupValue_Write    $00 | 0%
412  19.11.2022 17:45:42,026  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/83  EG Badezimmer Licht (Wand) Helligkeit Status  6  GroupValue_Write    $00 | 0%
434  19.11.2022 17:45:49,429  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer Licht PM 1  6  GroupValue_Write    $01 | Ein
435  19.11.2022 17:45:49,506  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $01 | Ein
436  19.11.2022 17:45:49,705  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $01 | Ein
438  19.11.2022 17:45:49,775  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $01 | Ein
443  19.11.2022 17:45:50,101  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $01 | Ein
447  19.11.2022 17:45:50,419  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/82  EG Badezimmer Licht PM 2 BW  6  GroupValue_Write    $01 | Ein
448  19.11.2022 17:45:50,788  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/73  EG Badezimmer Licht (Decke) Helligkeit Status  6  GroupValue_Write    $FF | 100%
449  19.11.2022 17:45:50,855  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/83  EG Badezimmer Licht (Wand) Helligkeit Status  6  GroupValue_Write    $FF | 100%
459  19.11.2022 17:45:52,239  vom Bus    Niedrig  1.0.142  Taster Badezimmer Flur  1/1/96  EG Badezimmer Licht Taster (Kurz)  6  GroupValue_Write    $00 | Aus
461  19.11.2022 17:45:52,344  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $00 | Aus
462  19.11.2022 17:45:52,447  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $00 | Aus
463  19.11.2022 17:45:52,500  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $00 | Aus
466  19.11.2022 17:45:52,811  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $00 | Aus
467  19.11.2022 17:45:53,021  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/83  EG Badezimmer Licht PM 2 TP  6  GroupValue_Write    $01 | Ein
468  19.11.2022 17:45:53,106  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/82  EG Badezimmer Licht PM 2 BW  6  GroupValue_Write    $00 | Aus
471  19.11.2022 17:45:53,436  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/7/80  EG Badezimmer Licht Reset (inv)  6  GroupValue_Write    $00 | Aus
473  19.11.2022 17:45:53,528  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer Licht PM 1  6  GroupValue_Write    $00 | Aus
474  19.11.2022 17:45:53,528  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/73  EG Badezimmer Licht (Decke) Helligkeit Status  6  GroupValue_Write    $00 | 0%
475  19.11.2022 17:45:53,583  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/83  EG Badezimmer Licht (Wand) Helligkeit Status  6  GroupValue_Write    $00 | 0%
476  19.11.2022 17:45:53,635  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $01 | Ein
477  19.11.2022 17:45:53,896  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $01 | Ein
478  19.11.2022 17:45:53,946  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $01 | Ein
482  19.11.2022 17:45:54,359  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $01 | Ein
484  19.11.2022 17:45:54,490  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer Licht PM 1  6  GroupValue_Write    $01 | Ein
[... cut ...]
435 VPM schaltet ein und Status wird versendet
447 PM 2 löst nun auch aus
459 Ich schalte direkt wieder aus
471 Reset welches an PM1 landet (TPM untersützt kein Reset ist aber auch nur zu Verlängerung und sendet selber auch keine weiter Telegramme innerhalb dieser Variante)
473 Jetzt kommt das aus vom PM 1 aufgrund des Reset
476 Schlatet direkt wieder ein.

Variante 3 (wie Variante 1 nur das Licht noch an ist)
484 PM 1 löst wieder aus
511 Warte bis Nachlauf von PM 1 raus ist.
517 Schalte manuell aus was diesmal auch klappt.