OpenKNX/OAM-PresenceModule #41: Fehler in der KNXProd und Busreset

traxanos, 2022-11-16:
Aktuell gibt es eine Situation die dazu führt, dass eine Element nach dem öffnen eines Projektes nicht mehr sichtbar ist:

Siehe auch: https://knx-user-forum.de/forum/projektforen/openknx/1808556-openknx-virtualpresence-release-vpm?p=1815853#post1815853

mumpf, 2022-11-17:
Hi, ich versuche das Setting rauszufinden, bei dem Raum Verlassen in der ETS verschwindet. 
Kriege ich nicht nicht. Weißt Du noch, was Du eingestellt hattest?

Gruß, Waldemar

traxanos, 2022-11-17:
Was verwendest du den ? ETS5 oder ETS6? Ich nutze die ETS6. Ich versuche das mal parallel nochmal zu reproduzieren.

mumpf, 2022-11-18:
Danke für das Beispielprojekt, ich konnte das Problem finden und lösen. 

Ist gelöst in Release 1.2

closed 2022-11-18 by mumpf