OpenKNX/OAM-PresenceModule #42: Licht schaltet sich nach einem Automatik-AUS direkt wieder ein.

traxanos, 2022-11-16:
Details dazu hier: https://knx-user-forum.de/forum/projektforen/openknx/1808556-openknx-virtualpresence-release-vpm?p=1815667#post1815667

Leider kann ich den Fehler nicht mehr reproduzieren, da ich die Konfig für mich zu weiter angepasst habe, so dass der Fehler nicht mehr auftritt. Aber existieren sollte trotzdem noch.

traxanos, 2022-11-18:
bzw. Auslöser/Grund dürfte vermutlich #41 sein.

mumpf, 2022-11-18:
Hi, nachdem ich jetzt die Ursache für #41 kenne, würde ich auch annehmen, dass es durch diesen Fehler kommt.

Vor allem dieses Statement aus Deiner Erklärung spricht dafür:

> Zuvor kommt nur das AUS und zeitlich sieht es so aus, als würde das AUS den VPM veranlassen alles einzuschalten.

Ich schließe dieses Issue auch, da ich es nach der Korrektur nicht mehr reproduzieren kann. Du kannst es gerne wiederöffnen, falls ich irgendwas übersehen habe.

Gruß, Waldemar 

(gelöst mit Release 1.2)

traxanos, 2022-11-18:
Meinst du nicht #40  statt #41?

mumpf, 2022-11-19:
Ja, ich meinte die #40. 

Das mit dem Test ist eine gute Idee, Du kannst den Branch "openknx-build" nutzen, für alle genutzten repos außer knx - der bleibt auf master.
Mit dem Dir schon bekannten "Build-Big" bekommst Du ein Setup-File im release-Verzeichnis.

Gruß, Waldemar

traxanos, 2022-11-19:
hab schon seit ner 1-2 stunde die 1.2 selber gebaut :D hab dazu aber nur die beiden commits mit cherry pick gezogen. dachte das müsste reichen. hab dann firmware aktuallisiert und auch die knxprod auf 1.2 hochgehoben.

hab aber immer noch das problem in einem anderen raum mit dem wiedereinschalten. muss aber jetzt kurz kaufen, dann kann ich nochmal schauen obs nicht was neues ist. oder doch noch ein konfig fehler...

traxanos, 2022-11-19:
So ich hab die Repos auf openbuild umgestellt und neu gebaut. Das Problem besteht weiter.

Wenn vom ersten PM 1 das Aus vor meinem Manuell-Aus (also Raum verlassen) kommt, klappt es. Läuft der PM 1 noch geht das licht wieder an.

traxanos, 2022-11-19:
Sorry hatte die Texte im falschen Ticket:

```
241  19.11.2022 13:39:16,208  vom Bus    Niedrig  1.0.142  Taster Badezimmer Flur  1/1/96  EG Badezimmer Licht Taster (Kurz)  6  GroupValue_Write    $00 | Aus
246  19.11.2022 13:39:16,360  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $00 | Aus
247  19.11.2022 13:39:16,360  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $00 | Aus
249  19.11.2022 13:39:16,406  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $00 | Aus
252  19.11.2022 13:39:16,713  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $00 | Aus
265  19.11.2022 13:39:17,443  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/7/80  EG Badezimmer Reset (inv)  6  GroupValue_Write    $00 | Aus
266  19.11.2022 13:39:17,489  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer PM 1  6  GroupValue_Write    $00 | Aus
267  19.11.2022 13:39:17,535  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $01 | Ein
269  19.11.2022 13:39:17,694  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $01 | Ein
270  19.11.2022 13:39:17,737  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $01 | Ein
273  19.11.2022 13:39:18,168  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $01 | Ein
282  19.11.2022 13:39:19,351  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/83  EG Badezimmer PM 2 TP  6  GroupValue_Write    $01 | Ein
284  19.11.2022 13:39:19,396  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/82  EG Badezimmer PM 2 BW  6  GroupValue_Write    $00 | Aus
```

Auf die schnelle ein Log Auszug (Konfig kann ich später schicken).

Hab mit 241 ausgeschaltet, der VPM schaltet mit 246 aus. Der BWM sendet in 266 ein AUS (aufgrund des PM Reset) und im Anschluss geht 267 geht das Licht wieder an. Dazwischen ist nicht gekommen.

Wenn also die beiden Cherrypicks reichen für die neue Version 1.2 (was mit auch die KNX für KNXProd + FW anzeigt), dann scheint es doch nicht behoben zu sein.

traxanos, 2022-11-19:
Hier ein größerer Auszug mit Erläuterung was passiert ist:

PM 1 ist der BWM an der Wand und als Bewegung verknüpft.
PM 2 (BW/TP als OR verknüpft) ist der Steinel TPM als "nur verlängern nicht einschalten"

```
Id Zeit  Dienst  Flags   Prio  Quelladresse  Quellname  Zieladresse  Zielname  Hop Count  Typ  DPT  Info
400  19.11.2022 17:45:40,659  vom Bus    Niedrig  1.0.142  Taster Badezimmer Flur  1/1/96  EG Badezimmer Licht Taster (Kurz)  6  GroupValue_Write    $00 | Aus
401  19.11.2022 17:45:40,709  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $00 | Aus
403  19.11.2022 17:45:40,879  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $00 | Aus
404  19.11.2022 17:45:40,879  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $00 | Aus
406  19.11.2022 17:45:41,202  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $00 | Aus
410  19.11.2022 17:45:41,795  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/7/80  EG Badezimmer Licht Reset (inv)  6  GroupValue_Write    $00 | Aus
411  19.11.2022 17:45:41,957  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/73  EG Badezimmer Licht (Decke) Helligkeit Status  6  GroupValue_Write    $00 | 0%
412  19.11.2022 17:45:42,026  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/83  EG Badezimmer Licht (Wand) Helligkeit Status  6  GroupValue_Write    $00 | 0%
434  19.11.2022 17:45:49,429  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer Licht PM 1  6  GroupValue_Write    $01 | Ein
435  19.11.2022 17:45:49,506  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $01 | Ein
436  19.11.2022 17:45:49,705  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $01 | Ein
438  19.11.2022 17:45:49,775  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $01 | Ein
443  19.11.2022 17:45:50,101  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $01 | Ein
447  19.11.2022 17:45:50,419  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/82  EG Badezimmer Licht PM 2 BW  6  GroupValue_Write    $01 | Ein
448  19.11.2022 17:45:50,788  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/73  EG Badezimmer Licht (Decke) Helligkeit Status  6  GroupValue_Write    $FF | 100%
449  19.11.2022 17:45:50,855  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/83  EG Badezimmer Licht (Wand) Helligkeit Status  6  GroupValue_Write    $FF | 100%
459  19.11.2022 17:45:52,239  vom Bus    Niedrig  1.0.142  Taster Badezimmer Flur  1/1/96  EG Badezimmer Licht Taster (Kurz)  6  GroupValue_Write    $00 | Aus
461  19.11.2022 17:45:52,344  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $00 | Aus
462  19.11.2022 17:45:52,447  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $00 | Aus
463  19.11.2022 17:45:52,500  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $00 | Aus
466  19.11.2022 17:45:52,811  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $00 | Aus
467  19.11.2022 17:45:53,021  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/83  EG Badezimmer Licht PM 2 TP  6  GroupValue_Write    $01 | Ein
468  19.11.2022 17:45:53,106  vom Bus    Niedrig  1.0.166  Präsenzmelder Badezimmer  1/7/82  EG Badezimmer Licht PM 2 BW  6  GroupValue_Write    $00 | Aus
471  19.11.2022 17:45:53,436  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/7/80  EG Badezimmer Licht Reset (inv)  6  GroupValue_Write    $00 | Aus
473  19.11.2022 17:45:53,528  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer Licht PM 1  6  GroupValue_Write    $00 | Aus
474  19.11.2022 17:45:53,528  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/73  EG Badezimmer Licht (Decke) Helligkeit Status  6  GroupValue_Write    $00 | 0%
475  19.11.2022 17:45:53,583  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/83  EG Badezimmer Licht (Wand) Helligkeit Status  6  GroupValue_Write    $00 | 0%
476  19.11.2022 17:45:53,635  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/90  EG Badezimmer Licht  6  GroupValue_Write    $01 | Ein
477  19.11.2022 17:45:53,896  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/71  EG Badezimmer Licht (Decke) Status  6  GroupValue_Write    $01 | Ein
478  19.11.2022 17:45:53,946  vom Bus    Niedrig  1.0.53  LED Controller EG 3  1/1/81  EG Badezimmer Licht (Wand) Status  6  GroupValue_Write    $01 | Ein
482  19.11.2022 17:45:54,359  vom Bus    Niedrig  1.0.201  OpenKNX: Präsenzmelder 1  1/1/91  EG Badezimmer Licht Status  6  GroupValue_Write    $01 | Ein
484  19.11.2022 17:45:54,490  vom Bus    Niedrig  1.0.169  Bewegungsmelder EG Badezimmer  1/7/81  EG Badezimmer Licht PM 1  6  GroupValue_Write    $01 | Ein
[... cut ...]
435 VPM schaltet ein und Status wird versendet
447 PM 2 löst nun auch aus
459 Ich schalte direkt wieder aus
471 Reset welches an PM1 landet (TPM untersützt kein Reset ist aber auch nur zu Verlängerung und sendet selber auch keine weiter Telegramme innerhalb dieser Variante)
473 Jetzt kommt das aus vom PM 1 aufgrund des Reset
476 Schlatet direkt wieder ein.

Variante 3 (wie Variante 1 nur das Licht noch an ist)
484 PM 1 löst wieder aus
511 Warte bis Nachlauf von PM 1 raus ist.
517 Schalte manuell aus was diesmal auch klappt.

mumpf, 2022-11-19:
Komisch, genau den Fall hatte ich getestet (glaube ich zumindest), aber klar, ich schau heute Abend nochmal.

mumpf, 2022-11-19:
Ich hoffe, ich hab es jetzt. Ich hab einen change gepushed, aber Vorsicht: Ist noch nicht von mir getestet. 

Falls Du (at your own Risk) testen willst, kannst Du es gerne machen, ich schreib hier auf jeden Fall nochmal rein, wenn meine Tests durch sind.

Ich hab auch gleich was für den Status Manuell gemacht.

Bei dem Setup-zip gibt es übrigens auch schon setup-files für den 1TE-RP2040-SmartMF vom Masifi.
Und für den OpenKNX-REG1-Base von Ing-Dom.
Beide noch ungetestet, da ich diese Hardware noch nicht habe.

Falls Du also Zeit und Lust hast, kannst Du gerne einiges ausprobieren :-) Mich würde auch interessieren, ob das Problem in der ETS mit dem verschwindenden LeaveRoom auch bei Dir weg ist, bei mir kommt es (auch mit Deinem Projekt) nicht mehr vor.

Gruß, Waldemar

traxanos, 2022-11-19:
Also ich teste die Ganze Zeit mit 1TE-RP2040-SmartMF Image. Bisher keine Auffäligkeiten.
Das Problem ist leider noch immer nicht behoben.

traxanos, 2022-11-19:
Und immer noch laut Monitor direkt wenn das Aus vom PM 1 nach dem Rest kommt. Spielt vielleicht doch noch der TPM mit rein? Dieser läßt sich nicht reseten. Aber eigentlich soll er nur den Status verlängern. Aber wertest du das immer als Ganzes aus. Wobei es als Nachlaufzeitverlänger eigentlich keine Relevanz haben dürfte.

mumpf, 2022-11-20:
[PresenceModule-Big-1.3.zip](https://github.com/OpenKNX/OAM-PresenceModule/files/10048182/PresenceModule-Big-1.3.zip)

So, ich hab jetzt noch alle versionen auf 1.3 hochgezogen, um zu verhindern, dass Du falsche Versionen hast. Und nutze bitte das beigefügte von mir gebaute File.
Deine Aussage,, dass Du noch immer 999 bei den Kanalnummern hast, zeigt mir, dass Du nicht richtig gebaut hast. 

Falls der Fehler immer noch existiert, bräuchte ich nochmal eine Kopie Deines Projektes (also nur der VPM). Und im Gruppenmonitor dann alle GA, die den VPM beeinflussen, also auch alle Status-GA, die vonaußen kommen, falls es noch zusätzliche gab.

Gute Nacht.

traxanos, 2022-11-20:
Danke dir.

Also ich hab nochmal nachgeschaut. Mit der 1.3 ist das 999er passe. Da ich aber von heute Nacht von die Console auf hatte und wissen wollte warum das mit einem Build nicht geklappt hat, hab ich nochmal meine History verglichen. Und ich kann mir das aktuell nur so erklären, dass du ein Push vergessen hast. Die letzte Version die mit beim Checkout angeboten wurde ist: adc4f4bb6cec49178b686b06ca958c2b64290297 (also die habe ich gebaut). Heute früh ein wiederholtes git pull lieferte mir: d1428071005515293557ccc89852d7814eff07c6.

Gestern 1.2
[image]

Heute 1.3
[image]

Damit fehlte mir der Commit 85b4abdb5b778d0eb36c47461f6e5ff2de8d5592 welches aber nur "Update documentation & Channel KO states are sent after Channel StartupDelay" anpasste.

Entsprechend müsste mein Fehler weiter existieren, was bei einem ersten Test auch weiterhin der Fall ist. Ich werde dir jetzt versuchen ein Demo-Projekt zu extrahieren, was ich aber nicht öffentlich posten möchte (wegen möglicher SecKeys). Ich werde dir eine PN per KNFUF schicken.

traxanos, 2022-11-20:
So Download ist per KNXUF raus und hier nochmal ein Protokoll

[image]

Hab das Demo-Projekt genommen, dann sieht man direkt was der VPM macht. Die GAs sind ja ordentlich beschriftet.

traxanos, 2022-11-20:
Ich hab es endlich gefunden!!! Es sind zwei verschiedene Bugs.

Der eigentliche Bug ist behoben. Denn als der Bug noch da war, konnte ich immer wieder ausschalten und das Licht ging auch immer wieder erneut an. Das ist seit dem letzten Update nicht mehr der Fall!

Grund ist wohl ein Timing-Bug im MDT BWM. Man wird ja beim Testen irgendwann betriebsblind. Immer wenn ich getestet habe, bin ich in den Raum rein / raus und Schalter für AUS gedrückt. Dabei sendet der VPM korrekt das Reset Z43 und schickt es an den BWM. Der BWM schickt darauf auch korrekterweise ein AUS Z44. Aber wenn dem BWM das alles zu schnell geht, schummelt sich doch noch ein EIN rein Z51.

Ich vermute der BWM von MDT wird inter auch einen Hardware-Status vom Sensor haben. Wenn das zu schnell geht, ist der Sensor vmtl. noch im "Bewegung erkannt"-Modus und beim nächsten Check glaubt er es wäre eine neue Bewegung.

Man muss auch ehrlich sein, wenn der vorherige Bug nicht vorhanden gewesen wäre, es wäre mir sonst nie aufgefallen. Wer läuft in einen Raum, geht  sofort wieder raus, schalten direkt aus + rennt dann direkt wieder rein.

PS: Ich hab mit der Todzeit gespielt. Ab einer Todzeit von 3s auf dem VPM + 1s auf dem BWM, ist der Fehler nicht mehr reproduzierbar. Selbst mit in dem Raum winken und direkt aus machen.

Sorry das ich dich damit so verrückt gemacht habe. Ich bin übrigens dafür ein Wiki mit "Best Practice" zu haben. Da könnte man solche Feinheiten dokumentieren. Wird im KNX Umfeld viel zu selten gemacht wie ich finde.

closed 2022-11-20 by traxanos

mumpf, 2022-11-20:
> Der eigentliche Bug ist behoben. Denn als der Bug noch da war, konnte ich immer wieder ausschalten und das Licht ging auch immer wieder erneut an. Das ist seit dem letzten Update nicht mehr der Fall!

Dann bin ich beruhigt, ich wusste nicht mehr, wo ich suchen sollte. Danke dass Du selber auch noch mal geschaut hast.

> PS: Ich hab mit der Todzeit gespielt. Ab einer Todzeit von 3s auf dem VPM + 1s auf dem BWM, ist der Fehler nicht mehr reproduzierbar. Selbst mit in den Raum winken und direkt aus machen.

Deswegen hab ich die Totzeit mit aufgenommen, damit man "Eigenheiten" des externen Melders überbrücken kann. 

> Sorry das ich dich damit so verrückt gemacht habe. Ich bin übrigens dafür ein Wiki mit "Best Practice" zu haben. Da könnte man solche Feinheiten dokumentieren. Wird im KNX Umfeld viel zu selten gemacht wie ich finde.

Nein, der Fehler war gut, ich habe wirklich keine Wiederholten AUS-Telegramme in den unterschiedlichen Situationen getestet, einfach nicht daran gedacht. Und Du hast letztendlich gleich 2 Situationen gefunden, die wirklich viele Leute gestört hätten. Für den MDT-Bug kann ich natürlich nichts :-)

Noch eine kurze Nachfrage: Du hast ja jetzt den 1TE-RP2040-SmartMF getestet. Damit läuft es also, oder? Ich hab die Hardware nämlich immer noch nicht :-)

traxanos, 2022-11-20:
Korrekt ich arbeite aktuell nur mit der 1TE-RP2040-SmartMF. Imho kann das Release-Public gehen.