OpenKNX/OAM-PresenceModule #43: Initiales senden von KOs

traxanos, 2022-11-19:
Was mir beim Testen immer wieder auffällt ist, dass nach einem Neustart keine initalen Werte übermittelt werden.

Zum Beispiel Modus. Habe ich vor dem programmieren den manuellen Modus aktiv, ist nach dem Neustart der Modus wieder auf Automatik. Aber das Sendet das Modul nicht. Daher glauben meine LEDs der Raum sei noch im Manuell-Modus.

Ich würde es gut finden, wenn bestimmte werte nach dem Neustart einmalig gesendet werden.

Modus & Sperre wären solche KOs. Ggf auch Nachlaufzeit (keine Ahnung nutze das nicht)

mumpf, 2022-11-19:
Können wir machen. Muss ich mir mal ansehen. Wird aber nichts mehr zum heutigen Release. Und ist auch nicht ganz so schnell zu machen, da ich nicht für 40 Kanäle direkt nach dem Neustart für viele KO initiale Werte senden kann. 

Workaround: Lasse das Logikmodul nach dem Neustart ein Read-Request senden, dessen GA Du als hörende GA an die KO verknüpfst, die senden sollen. Dann hast Du sofort eine Lösung.

traxanos, 2022-11-19:
ja das hab ich schon im einsatz für was anderes. für "Modus" lohnt das nicht. wenn das irgendwan also optimierung kommt reicht das. idr sind die kanäle nicht im manuell modus. also sehr selten de fall bei mir :D

willisurf, 2022-11-19:
Würde mir auch gut gefallen. Wenn das an der wählbaren Kanalverzögerung hängt, entzerrt sich das zeitlich ja etwas.

mumpf, 2022-11-19:
Hi,

nochmal nachgedacht: Nachlaufzeit und Helligkeitsschwelle für Ein- und Ausschalten werden jetzt schon gesendet, wenn man das Ü-Flag setzt. Ich habe mich dafür entschieden, das so zu machen, da ich glaube, dass das selten gebraucht wird und ich dafür nicht die Applikation komplizierter machen muss. Sperre ist auch so gelöst (da hab ich jetzt nicht nachgeschaut, bin mir aber ziemlich sicher).

Und Modus muss ich noch nachziehen, da gibt es noch nichts.

Gruß, Waldemar

mumpf, 2022-11-20:
Hi, 

mit 1.2.2 (jetzt 1.3) habe ich das Senden vom Status bei Neustart an den Status des StartupDelay des Kanals gekoppelt. Dabei auch noch das Senden des Manuell-Modus hinzugefügt. 

War einfacher als gedacht, da ich schon einen Zustandsübergang von StartupDelay zu Running hatte.

Gruß, Waldemar

closed 2022-11-20 by mumpf