OpenKNX/OAM-PresenceModule #44: 999er Limit bei KO-Nummeranzeige

traxanos, 2022-11-19:
Zur Vollständigkeit, es gibt noch den Bug, dass die KO Nummern in der KNXProd bei 999 enden.

https://knx-user-forum.de/forum/projektforen/openknx/1765082-openknx-logikmodul-release?p=1818271#post1818271

mumpf, 2022-11-19:
Das ist bereits mit Release 1.2 gelöst.

closed 2022-11-19 by mumpf

traxanos, 2022-11-19:
Ähm. Nicht bei mir. Habe laut KNX die WP-Presence-Logic V1.2 drauf aber die höheren Zahlen sind alle 999 in der UI. hattest du nicht im forum geschrieben das du OpenKNXproducer anpassen musst dafür. ich sehe auch keine änderung hier.

mumpf, 2022-11-20:
Hi, nur zur Erklärung:

Wenn Du selber baust, kannst Du eigentlich nicht mehr die offiziellen Release-Nummern nehmen. Deswegen liefert mein Build standardmäßig eine neue (inkompatible) devel-Applikation.

Hintergrund:
Meine Applikationen bestehen aus mehreren unabhängigen Modulen und müssen passend compiliert und produziert werden. Die ETS zeigt immer nur die Version des "Hauptmoduls", aber die enthlatenen Module müssen auch die passenden Versionen haben. 

Der Fix mit den 999 war im Logikmodul (commit f4c6d92). Danach muss man den VPM (Firmware und ETS) neu bauen. Hast Du vorher schon die 1.2 vom VPM mit einem Cherrypick geholt, passt die ETS-Versionierung nicht mehr. Das ist sehr aufwändig und mit großer Vorsicht zu genießen. Ist auch der Grund, warum ich die Version auf 1.3 hochgezogen  habe, als Dank für den Test. Aber das werden wir nicht beiliebig so treiben können, dann gehen mir zu schnell die ETS-Versionsnummern aus.

Das Build-Big-Script ist eigentlich ein Produktionsscript, der ganz am Ende, sozusagen für das Release, nur aufgerufen wird. Falls Du öfter Testen willst, müssen wir uns was anderes Überlegen.

P.S.: Für das ETS-Versionsschema kann ich nichts, ich habe versucht, das Beste draus zu machen.

traxanos, 2022-11-20:
Kann man nicht mit Buchstaben wie 1.2b arbeiten. MDT nutzt glaube ich auch Buchstaben.

willisurf, 2022-11-20:
Die KO Nummer direkt ist schon sinnvoll. Ich nutze das als Hinweis für direkte intere Verknüpfungen. Da ist es dann wichtig bei der Systematik der KO Numerierung der ETS zu bleiben.

traxanos, 2022-11-20:
genau. aber man erklärte mir schon die versionierung der ets :D