OpenKNX/OAM-PresenceModule #48: Szene schaltet nach Ablauf der Rückfallzeit nicht aus

ginkel, 2023-01-03:
Moin,

ich habe hier einen möglichen Bug im VPM gefunden. Aktuell im Einsatz: v1.5, war aber auch bei der Vorversion schon so.

Und zwar habe ich im Bad einen VPM als Präsenzmelder konfiguriert, den ich mittels KO zum Duschen sperren kann (da der PM nicht durch die Duschwand sehen kann; Rückfall nach 10 Minuten):

[image]

[image]

[image]

Aktuell ist Tag, also hänge ich nur die Config dieser Phase an:

[image]

[image]

Hier noch die verbundenen KOs:

[image]

Beim Verlassen des Raums löse ich via Taster die Szene 30 aus. Wenn nun zuvor die Sperre aktiviert war und die Rückfallzeit abgelaufen ist, lässt sich das Licht nicht per Taster über die Szene ausschalten, die Szene hat im VPM also keine Wirkung:

[image]

Wurde die Sperre hingegen ohne Ablauf der Rückfallzeit deaktiviert, schaltet der VPM erwartungskonform (Telegramm 2592):

[image]

Analog wenn keine Sperre aktiv war:

[image]

Ideen?

Danke & viele Grüße
Thilo

mumpf, 2023-01-03:
Hi,

erstmal danke für das Feedback. Ich muss das mal nachstellen. Als Workaround / Vereinfachung würde ich gar nicht sperren. Schalte einfach Manuell EIN statt Sperre und über Szene 1 bzw. 30 nur Raum Verslassen. Mehr braucht es für Deinen Fall nicht - glaube ich zumindest....

Den Rest schaue ich mir noch an, bin aber derzeit im Urlaub, deswegen geht das nicht sofort.

Gruß,  Waldemar

mumpf, 2023-01-03:
P.S.: Ich bin fest überzeugt, dass man für die meisten PM-Szenarien keine Sperren braucht... Eigentlich will man immer nur manuell eingreifen - wie auch bei Dir - und dafür gibt es den Manuell-Modul.