OpenKNX/OAM-PresenceModule #49: Präsenzkanal (Halbautomat) schaltet beim Wechsel der Tagesphase das Licht aus

ginkel, 2023-01-30:
Moin,

ich habe einen VPM-Kanal auf Basis der v1.5.3 als Halbautomat zum Ausschalten des Lichts nach Ende bei mehr als 15 Minuten ohne Präsenz konfiguriert. AFAICS schaltet der Kanal aber auch beim Wechsel der Tagesphase das Licht aus, obwohl ununterbrochen Präsenz gegeben ist. Der Kanal ist wie folgt parametriert:

[image]

[image]

Die einzelnen Tagesphasen sind alle identisch, daher hier nur die erste:

[image]

Hier noch die verknüpften KOs:

[image]

Aus dem Log:

```
2023-01-30 19:00:01.191 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="2/1/6" payload="<GroupValueWrite value="<DPTArray value="[0x1]" />" />" />
2023-01-30 19:00:01.285 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="1/3/3" payload="<GroupValueWrite value="<DPTArray value="[0x0]" />" />" />
2023-01-30 19:00:01.312 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="2/0/8" payload="<GroupValueWrite value="<DPTBinary value="0" />" />" />
2023-01-30 19:00:01.343 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="1/3/8" payload="<GroupValueWrite value="<DPTArray value="[0x0]" />" />" />
2023-01-30 19:00:01.375 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="1/3/15" payload="<GroupValueWrite value="<DPTArray value="[0x0]" />" />" />
2023-01-30 19:00:01.407 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="1/3/1" payload="<GroupValueWrite value="<DPTArray value="[0x0]" />" />" />
2023-01-30 19:00:01.437 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="0/1/8" payload="<GroupValueWrite value="<DPTBinary value="0" />" />" />
2023-01-30 19:00:01.466 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="0/4/11" payload="<GroupValueWrite value="<DPTBinary value="0" />" />" />
2023-01-30 19:00:01.505 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="2/1/0" payload="<GroupValueWrite value="<DPTBinary value="0" />" />" />
2023-01-30 19:00:01.535 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="2/1/3" payload="<GroupValueWrite value="<DPTBinary value="0" />" />" />
2023-01-30 19:00:01.619 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.light_livingroom_floorlamp
2023-01-30 19:00:01.623 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.light_livingroom_floorlamp
2023-01-30 19:00:01.630 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Outgoing" source_address="1.1.42" destination_address="0/2/17" payload="<GroupValueWrite value="<DPTBinary value="False" />" />" />
2023-01-30 19:00:01.647 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Incoming" source_address="1.1.2" destination_address="0/1/11" payload="<GroupValueWrite value="<DPTBinary value="0" />" />" />
```

2/1/6 ist die GA der Tagesphase, 0/4/11 schaltet dann das Licht im Wohnzimmer aus. 1.1.2 ist die PA des VPM.

Ideen?

Merci,
Thilo

mumpf, 2023-01-30:
Hi Thilo,

nur um das einzugrenzen: Passiert das auch beim Vollautomaten? Alternativ: Stell mal ein, dass er die Tagesphase sofort wechseln soll und nicht erst nach dem Ausschalten.

An sich sagst Du in Deinen Einstellungen ja schon, dass er die Tagesphase erst wechseln soll, wenn das Licht aus geht, nur sollte es natürlich nicht sofort ausgehen sondern noch die Nachlaufzeit abwarten.

Ansonsten: Warum 4 identische Tagesphasen? Wenn das so bleiben soll, dann mach stattessen nur eine und keinen Tagesphasenwechsel, als Workaround.

Gruß, Waldemar

ginkel, 2023-01-31:
Hallo Waldemar,

danke für die schnelle Rückmeldung! 

Ich habe ein paar Kombinationen ausprobiert:

- Sofortiger Wechsel der Tagesphase ändert das Verhalten nicht.
- Umstellung auf Vollautomat (auch mit "nur AUS senden") hilft, sofern beim oder nach dem Einschalten des Lichts Präsenz gegeben ist. Bei manuellem Schalten ohne Präsenz (z.B. aus der ETS vom Schreibtisch aus) geht das Licht beim Tagesphasenwechsel weiterhin aus.

Die vier Tagesphasen habe ich wie bei meinen anderen PM-Kanälen eingerichtet, die tatsächlich ein anderes Schaltverhalten je Tagesphase haben. Insofern ist die Differenzierung aktuell tatsächlich unnötig - hat eher was von "reserved for future use".

Viele Grüße
Thilo

P.S.: Hoffe, die Beschreibung ist soweit klar. Musste ich beim Testen weiterer Kombinationen mehrfach umbauen, da meine Hypothesen nicht mehr passten.

mumpf, 2023-03-02:
Hi Thilo,

bin am Abarbeiten der Bugs beim VPM. Das Release 1.7.7 wird einen Fix enthalten, der ein gleiches Verhalten beim Vollautomat und beim Halbautomat bezüglich Präsenz haben wird.
Das Adressiert den folgenden Fall aus Deinem Test mit Vollautomaten von oben:

> Umstellung auf Vollautomat (auch mit "nur AUS senden") hilft, sofern beim oder nach dem Einschalten des Lichts Präsenz gegeben ist.

Ist jetzt auch beim Halbautomaten so.

> Bei manuellem Schalten ohne Präsenz (z.B. aus der ETS vom Schreibtisch aus) geht das Licht beim Tagesphasenwechsel weiterhin aus.

Du sagst dem VPM aber auch nicht, dass Du manuell schaltest! Bei Dir sind die beiden KO 133/134 nicht belegt. Dann kann der VPM auch nicht wissen, dass der Schaltvorgang beabsichtigt war. Du solltest das Schaltsignal an einen der beiden Eingänge schicken, je nachdem, wie Du den Präsenzmelder durch externes schalten übersteuern willst (Siehe Applikationsbeschreibung).

Ich gehe fest davon aus, dass es dann wie gewünscht funktioniert.

Gruß, Waldemar

closed 2023-03-02 by mumpf

ginkel, 2023-03-02:
Hallo Waldemar,

danke für dein Feedback! In vielen Räumen kann ich das tatsächlich so umsetzen, da der VPM alle Leuchten homogen schaltet. Im Wohnzimmer haben wir aber zwei Lampengruppen, die manuell ein aber automatisch aus geschaltet werden sollen. Verbinde ich dann einfach die verODERte GAen zum Schalten der Lampengruppen dann mit dem VPM oder hast du eine bessere Idee?

Merci,
Thilo